m2-farzan/ros2-galactic-PKGBUILD #1: [mimick_vendor] _FORTIFY_SOURCE requires compiling with optimization (-O)

AchmadFathoni, 2021-06-27:
```
--- stderr: mimick_vendor                                                               
Cloning into 'mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039'...
HEAD is now at f171450 Add armv7l as a 32-bit ARM architecture. (#16)
In file included from /usr/include/errno.h:25,
                 from /home/toni/.cache/yay/ros2-galactic/src/build/mimick_vendor/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039-prefix/src/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039/include/mimick/mock.h:27,
                 from /home/toni/.cache/yay/ros2-galactic/src/build/mimick_vendor/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039-prefix/src/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039/include/mimick/mimick.h:401,
                 from /home/toni/.cache/yay/ros2-galactic/src/build/mimick_vendor/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039-prefix/src/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039/test/test.c:1:
/usr/include/features.h:397:4: error: #warning _FORTIFY_SOURCE requires compiling with optimization (-O) [-Werror=cpp]
  397 | #  warning _FORTIFY_SOURCE requires compiling with optimization (-O)
      |    ^~~~~~~
In file included from /usr/include/c++/11.1.0/x86_64-pc-linux-gnu/bits/os_defines.h:39,
                 from /usr/include/c++/11.1.0/x86_64-pc-linux-gnu/bits/c++config.h:571,
                 from /usr/include/c++/11.1.0/cstdarg:42,
                 from /home/toni/.cache/yay/ros2-galactic/src/build/mimick_vendor/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039-prefix/src/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039/include/mimick/literal.h:31,
                 from /home/toni/.cache/yay/ros2-galactic/src/build/mimick_vendor/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039-prefix/src/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039/include/mimick/mimick.h:29,
                 from /home/toni/.cache/yay/ros2-galactic/src/build/mimick_vendor/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039-prefix/src/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039/test/test.cpp:1:
/usr/include/features.h:397:4: warning: #warning _FORTIFY_SOURCE requires compiling with optimization (-O) [-Wcpp]
  397 | #  warning _FORTIFY_SOURCE requires compiling with optimization (-O)
      |    ^~~~~~~
In file included from /usr/include/errno.h:25,
                 from /home/toni/.cache/yay/ros2-galactic/src/build/mimick_vendor/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039-prefix/src/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039/include/mimick/mock.h:27,
                 from /home/toni/.cache/yay/ros2-galactic/src/build/mimick_vendor/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039-prefix/src/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039/include/mimick/mimick.h:401,
                 from /home/toni/.cache/yay/ros2-galactic/src/build/mimick_vendor/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039-prefix/src/mimick-f171450b5ebaa3d2538c762a059dfc6ab7a01039/sample/strdup/test.c:1:
/usr/include/features.h:397:4: error: #warning _FORTIFY_SOURCE requires compiling with optimizat
```

m2-farzan, 2021-06-27:
Hi,

Thanks for reporting. I have fixed the package.

Note that you may need to perform a clean build. If you're using `pkgbuild`, just start in a new directory. If you're using `yay`, you should remove the cache cache that should be in a directory like `~/.cache/yay/ros2-galactic`.

Please close the issue after verifying the error is gone.

AchmadFathoni, 2021-06-27:
Even after I `rm -rf ~/.cache/yay/ros2-galactic` the error is still there

m2-farzan, 2021-06-27:
Weird... I just ran a clean pkgbuild and it worked fine.

Can you try to build and install the package without yay? Use the following commands:
```
git clone https://aur.archlinux.org/ros2-galactic.git
cd ros2-galactic
makepkg -i
```

AchmadFathoni, 2021-06-27:
Still there, do you think my `/etc/makepkg.conf` contributes to this problem?
```
#!/hint/bash
#
# /etc/makepkg.conf
#

#########################################################################
# SOURCE ACQUISITION
#########################################################################
#
#-- The download utilities that makepkg should use to acquire sources
#  Format: 'protocol::agent'
DLAGENTS=('file::/usr/bin/curl -gqC - -o %o %u'
          'ftp::/usr/bin/curl -gqfC - --ftp-pasv --retry 3 --retry-delay 3 -o %o %u'
          'http::/usr/bin/curl -gqb "" -fLC - --retry 3 --retry-delay 3 -o %o %u'
          'https::/usr/bin/curl -gqb "" -fLC - --retry 3 --retry-delay 3 -o %o %u'
          'rsync::/usr/bin/rsync --no-motd -z %u %o'
          'scp::/usr/bin/scp -C %u %o')

# Other common tools:
# /usr/bin/snarf
# /usr/bin/lftpget -c
# /usr/bin/wget

#-- The package required by makepkg to download VCS sources
#  Format: 'protocol::package'
VCSCLIENTS=('bzr::bzr'
            'fossil::fossil'
            'git::git'
            'hg::mercurial'
            'svn::subversion')

#########################################################################
# ARCHITECTURE, COMPILE FLAGS
#########################################################################
#
CARCH="x86_64"
CHOST="x86_64-pc-linux-gnu"

#-- Compiler and Linker Flags
#CPPFLAGS=""
CFLAGS="-march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions \
        -Wp,-D_FORTIFY_SOURCE=2 -Wformat \
        -fstack-clash-protection -fcf-protection"
CXXFLAGS="$CFLAGS -Wp,-D_GLIBCXX_ASSERTIONS"
LDFLAGS="-Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now"
#RUSTFLAGS="-C opt-level=2"
#-- Make Flags: change this for DistCC/SMP systems
MAKEFLAGS="-j8"
#-- Debugging flags
DEBUG_CFLAGS="-g -fvar-tracking-assignments"
DEBUG_CXXFLAGS="-g -fvar-tracking-assignments"
#DEBUG_RUSTFLAGS="-C debuginfo=2"

#########################################################################
# BUILD ENVIRONMENT
#########################################################################
#
# Makepkg defaults: BUILDENV=(!distcc !color !ccache check !sign)
#  A negated environment option will do the opposite of the comments below.
#
#-- distcc:   Use the Distributed C/C++/ObjC compiler
#-- color:    Colorize output messages
#-- ccache:   Use ccache to cache compilation
#-- check:    Run the check() function if present in the PKGBUILD
#-- sign:     Generate PGP signature file
#
BUILDENV=(!distcc color !ccache check !sign)
#
[... 66 lines cut ...]
#
COMPRESSGZ=(gzip -c -f -n)
COMPRESSBZ2=(bzip2 -c -f)
COMPRESSXZ=(xz -c -z -)
COMPRESSZST=(zstd -c -z -q -)
COMPRESSLRZ=(lrzip -q)
COMPRESSLZO=(lzop -q)
COMPRESSZ=(compress -c -f)
COMPRESSLZ4=(lz4 -q)
COMPRESSLZ=(lzip -c -f)

#########################################################################
# EXTENSION DEFAULTS
#########################################################################
#
PKGEXT='.pkg.tar.zst'
SRCEXT='.src.tar.gz'

#########################################################################
# OTHER
#########################################################################
#
#-- Command used to run pacman as root, instead of trying sudo and su
#PACMAN_AUTH=()
```

m2-farzan, 2021-06-27:
Yes. Good. The problem probably lies at that line with `-D_FORTIFY_SOURCE=2` in `CFLAGS` section of the file. Remove the `-D_FORTIFY_SOURCE=2` and try again. (Don't forget to clean build)

AchmadFathoni, 2021-06-27:
Hmm, rather than changing global `/etc/makepkg.conf`, would it be better to put `unset CFLAG` in build()? Why is your PKGBUILD using`unset CPPFLAGS`?

m2-farzan, 2021-06-27:
You're right. It seems that pacman has changed the default `/etc/makepkg.conf` at some point between pacman 5.2.2 and 6.0.0. In the older version (which I happened to have), `-D_FORTIFY_SOURCE=2` was at `CPPFLAGS`, hence the `unset CPPFLAGS` line. I thought your conf file was modified but I was wrong.

m2-farzan, 2021-06-27:
Hope this fixes the issue.

closed 2021-06-27 by AchmadFathoni