m2-farzan/ros2-galactic-PKGBUILD #2: [ros1_bridge] error: non-constant condition for static assertion

AchmadFathoni, 2021-06-27:
```
/home/toni/.cache/yay/ros2-galactic/src/build/ros1_bridge/generated/geometry_msgs__msg__AccelWithCovariance__factories.cpp:69:34: error: non-constant condition for static assertion
   69 |     (ros2_msg.covariance.size()) >= (ros1_msg.covariance.size()),
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/toni/.cache/yay/ros2-galactic/src/build/ros1_bridge/generated/geometry_msgs__msg__AccelWithCovariance__factories.cpp:69:34: error: ‘ros2_msg’ is not a constant expression
/home/toni/.cache/yay/ros2-galactic/src/build/ros1_bridge/generated/geometry_msgs__msg__AccelWithCovariance__factories.cpp: In static member function ‘static void ros1_bridge::Factory<ROS1_T, ROS2_T>::convert_2_to_1(const ROS2_T&, ROS1_T&) [with ROS1_T = geometry_msgs::AccelWithCovariance_<std::allocator<void> >; ROS2_T = geometry_msgs::msg::AccelWithCovariance_<std::allocator<void> >]’:
/home/toni/.cache/yay/ros2-galactic/src/build/ros1_bridge/generated/geometry_msgs__msg__AccelWithCovariance__factories.cpp:100:34: error: non-constant condition for static assertion
  100 |     (ros1_msg.covariance.size()) >= (ros2_msg.covariance.size()),
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

m2-farzan, 2021-06-27:
We had a similar problem with the Foxy package. I don't have ROS 1 to reproduce this error, so can you please test this fix [1] and report if it works so that I can add it to the package?
[1] https://github.com/m2-farzan/ros2-foxy-PKGBUILD/commit/89f94a2a39f33ca68887c3defeb0091d060b7fca

AchmadFathoni, 2021-06-27:
Yes, that patch works.

closed 2021-06-27 by AchmadFathoni