m2-farzan/ros2-galactic-PKGBUILD #8: rviz_rendering: variable NOTFOUND (library_abs)

oysstu, 2021-12-13:
Getting the following error when recompiling for python 3.10. Not sure if it is due to newer version of cmake, assimp, or ros2.

```--- stderr: rviz_rendering                                                                             
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
library_abs
    linked by target "rviz_rendering" in directory <...>/ros2-galactic/src/ros2/src/ros2/rviz/rviz_rendering

CMake Generate step failed.  Build files cannot be regenerated correctly.
```
The variable is defined as a temporary variable in rviz_assimp_vendor, and is not used directly in the rviz_rendering CMakeLists.txt, but CMake still throws an error (possibly due to the ament_target_dependencies(PUBLIC rviz_assimp_vendor) macro.

m2-farzan, 2021-12-14:
Thanks for reporting the issue. I'm on a tight schedule right now, so I'll greatly appreciate a PR. By the way, did you try to clean-build?

oysstu, 2021-12-14:
Hi, I think this is an upstream issue. I'm not quite sure how to fix it, since the variable is declared in another package than where the error occurs.

oysstu, 2021-12-14:
Looks like an issue has been created at already at https://github.com/ros2/ros2/issues/1222. Closing this, as this hopefully will be resolved upstream.

closed 2021-12-14 by oysstu

m2-farzan, 2021-12-29:
Patched by 343c9f3a2505b2e50ca4dacf3f96de67984d703e. Thanks!