m2-farzan/ros2-galactic-PKGBUILD #9: python-pyqt5 dropped sip4-support patch

0ljik, 2021-12-25:
python-pyqt5 dropped sip4-support patch, so there is problem building qt_gui_cpp.

m2-farzan, 2021-12-25:
Yes. I don't know how to fix that but a temporary workaround is to downgrade python-pyqt5:
```
sudo pacman -U /var/cache/pacman/pkg/python-pyqt5-5.15.5-1-x86_64.pkg.tar.zst
```
If the cached package is missing, download it from here: https://archive.archlinux.org/packages/p/python-pyqt5/python-pyqt5-5.15.5-1-x86_64.pkg.tar.zst

m2-farzan, 2021-12-25:
Relevant: https://www.riverbankcomputing.com/pipermail/pyqt/2021-November/044346.html

m2-farzan, 2022-01-08:
Here's an alternative solution:
```
$ pip3 install pyqt5==5.15.5
```

This is the error message (for future visitors):
```
--- stderr: qt_gui_cpp                                                                                                                                    
<string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
/home/mostafa/tmp/ros2-rolling/src/install/share/python_qt_binding/cmake/sip_configure.py:2: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.spawn import find_executable
sip: /usr/lib/python3.10/site-packages/PyQt5/bindings/QtCore/QtCoremod.sip:23: syntax error
Traceback (most recent call last):
  File "/home/mostafa/tmp/ros2-rolling/src/install/share/python_qt_binding/cmake/sip_configure.py", line 123, in <module>
    subprocess.check_call(cmd)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/sip', '-c', '/home/mostafa/tmp/ros2-rolling/src/build/qt_gui_cpp/sip/qt_gui_cpp_sip', '-b', '/home/mostafa/tmp/ros2-rolling/src/build/qt_gui_cpp/sip/qt_gui_cpp_sip/pyqtscripting.sbf', '-I', '/usr/lib/python3.10/site-packages/PyQt5/bindings', '-w', '-n', 'PyQt5.sip', '-t', 'Qt_5_15_0', '-t', 'WS_X11', 'qt_gui_cpp.sip']' returned non-zero exit status 1.
make[2]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/build.make:103: sip/qt_gui_cpp_sip/Makefile] Error 1
make[1]: *** [CMakeFiles/Makefile2:279: src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< qt_gui_cpp [35.5s, exited with code 2]
```

We may need to take this discussion to upstream: https://github.com/ros-visualization/qt_gui_core

Some potentially useful info here: https://groups.google.com/g/linux.debian.bugs.dist/c/FdgUVpZ4SWE

goekce, 2022-01-09:
> Yes. I don't know how to fix that but a temporary workaround is to downgrade python-pyqt5:
>
> ```
> sudo pacman -U /var/cache/pacman/pkg/python-pyqt5-5.15.5-1-x86_64.pkg.tar.zst
> ```
>
> If the cached [...]

Downgrading won't work, because the package is based on python-3.9 :(

`pip` or via `pipman` is probably a better option.

goekce, 2022-01-14:
Update:

I did not want to overwrite my Python distribution using `sudo pip` so I tried using `pipman` and `pypi2pkgbuild.py` to create a package for `python-pyqt5` 5.15.5. Unfortunately `pipman` and [`pypi2pkgbuild.py` does not seem to support installing a specific version](https://github.com/anntzer/pypi2pkgbuild/issues/21) :(

I find this workaround that you pointed out the best for installation, but did not try if ROS breaks or not during development:
>  Also related (from https://aur.archlinux.org/packages/ros2-foxy/#comment-845555):

>>    I was able to workaround the issue by changing line 23 in /usr/share/sip/PyQt5/QtCore/QtCoremod.sip to %Module PyQt5.QtCore

goekce, 2022-03-06:
More info:

https://github.com/ros-noetic-arch/ros-noetic-rviz/issues/4

JorgeEdGG, 2022-03-10:
The problem is the sip4 support so
I resolve the issue by: 

Using the software manager:
Remove any ROS2 packages
Remove any pyqt5 packages
(don`t worry the next packages will install them latter)

Install python-pyqt5-sip4-compat (this will install other required pyqt5 packages)
Install ros2-galatic

m2-farzan, 2022-03-11:
Thanks @JorgeEdGG for sharing your solution. Unfortunately I can't find "python-pyqt5-sip4-compat" in AUR or Arch packages page. I've only found [this Manjaro package](https://software.manjaro.org/package/python-pyqt5-sip4-compat). I suppose your solution can only be used by Manjaro users. Am I right?

JorgeEdGG, 2022-03-11:
@m2-farzan You are right, that`s the package I installed. I`m using Manjaro but not sure if is only for this distro, maybe you can add the extra repo.

kistlin, 2022-06-07:
If you are on `Arch` you can use [python-pyqt5-sip4](https://aur.archlinux.org/packages/python-pyqt5-sip4) from the `AUR`.
After installing that and replacing the official package, the compilation/installation of `ros2-galactic` succeeded.

If the official package has it fixed, might someone notify me? Thanks :).

mr-cn, 2023-01-19:
Is sip4 supported version of PyQT5 is needed only at compiling time? or is it also needed in runtime?

mr-cn, 2023-01-19:
Both `python-pyqt5-sip4-compat` way or upgrading `sip4` way mentioned by nickalcock in https://github.com/ros-visualization/qt_gui_core/issues/249#issuecomment-800212338 can not solve this in my arch.
I successfully solve it by manually installing `python-pyqt5-sip4` package replacing the original `python-pyqt5` package.

m2-farzan, 2023-01-20:
@mr-cn do you have the time to make a PR for the workaround you've found? Does it suffice to simply to replace `ros2-pyqt5-sip-compat`, `sip4`, and `ros2-pyqt5-sip-compat` (from ros2-arch-deps package) with `python-pyqt5-sip4`?

ruffson, 2023-01-27:
@m2-farzan Can you please have a look at my PR addressing this issue when you have time?

m2-farzan, 2023-01-27:
Sure. I'll merge it in a few days. Sorry, it's been a busy week...

mr-cn, 2023-02-14:
Sorry for my late response. 

> Does it suffice to simply to replace ros2-pyqt5-sip-compat, sip4, and ros2-pyqt5-sip-compat (from ros2-arch-deps package) with python-pyqt5-sip4?

I suggest `sip4` package may not suppose to be removed, since `python-pyqt5-sip4` package only re-enables the pyqt5 side support for sip4, not providing sip4 package. 

We may move `python-pyqt5-sip4` into build time dependencies. This can avoid replacing system `python-pyqt5` package.

I will do some tests and share the result later.

GloriousEggroll, 2024-02-29:
There is a patch here for sip4 which backports upstream sip module fixes rather than re-adding a previously dropped python patch:

https://raw.githubusercontent.com/robwoolley/meta-openembedded/ad1494b75f6ece452bdaec88c32772a32cd9186b/meta-oe/recipes-devtools/sip/sip3/added-the-py_ssize_t_clean-argument-to-the-module-directive.patch

I rebased this on the latest 4.19.25 used in RHEL and it appears to work + be a better approach. (going forward with sip via backport instead of backwards with python-qt5 by re-adding a dropped patch).

Both of these patches would be needed on top of sip 4.19.25:
(1) https://copr-dist-git.fedorainfracloud.org/cgit/tcrider/autosd-ros1/sip.git/tree/sip-4.19.25-py_ssize_t_clean.patch?h=centos-stream%2bepel-next8
(2) https://copr-dist-git.fedorainfracloud.org/cgit/tcrider/autosd-ros1/sip.git/tree/sip-4.19.25-py_ssize_t_clean_module_directive.patch?h=centos-stream%2bepel-next8