m2-farzan/ros2-galactic-PKGBUILD #11: ln: failed to create symbolic link '/home/joe/.cache/paru/clone/ros2-pyqt5-sip-compat/pkg/ros2-pyqt5-sip-compat/usr/share/sip/PyQt5/./*':

KenwoodFox, 2022-01-02:
Im getting the below error when attempting to build

```
ln: failed to create symbolic link '/home/joe/.cache/paru/clone/ros2-pyqt5-sip-compat/pkg/ros2-pyqt5-sip-compat/usr/share/sip/PyQt5/./*': File exists
==> ERROR: A failure occurred in package().
    Aborting...
error: failed to build 'ros2-pyqt5-sip-compat-0.2-1':
```

Is this maybe linked to using paru?

KenwoodFox, 2022-01-02:
yay has a similar issue so i dont think its helper specific
```
ln: failed to create symbolic link '/home/joe/.cache/yay/ros2-pyqt5-sip-compat/pkg/ros2-pyqt5-sip-compat/usr/share/sip/PyQt5/./*': File exists
==> ERROR: A failure occurred in package().
    Aborting...
 -> error making: ros2-pyqt5-sip-compat
```

m2-farzan, 2022-01-03:
Hi, this error is probably because the target symlinks already exist. Study PKGBUILD of ros2-pyqt5-sip-compat to see which existing symlinks are problematic and then remove them. Alternatively, remove ros2-pyqt5-sip-compat from ros2-galactic dependencies by commenting the corresponding line in the PKGBUILD (this approach would require a manual makepkg). The issue is probably linked to your previous attempt at installing foxy (because in foxy, the symlinks are generated as a by-product of package preparation, which was an anti-pattern that we've fixed now, but unfortunately causes migration problems like this one).

KenwoodFox, 2022-01-06:
> Hi, this error is probably because the target symlinks already exist. Study PKGBUILD of ros2-pyqt5-sip-compat to see which existing symlinks are problematic and then remove them. Alternatively, [...]

Its good to hear its fixed now! very interesting though.. guess ive got some fixing to do, will this happen again if i move from glactic to what comes next?..

KenwoodFox, 2022-01-06:
Interesting im getting a new error, could be linked to the recent arch linux python 3 thing
```
==> Making package: ros2-galactic 2021.07.16-5 (Thu 06 Jan 2022 02:20:17 AM EST)
==> Retrieving sources...
  -> Updating ros2 git repo...
Fetching origin
  -> Found google_benchmark_vendor.patch
  -> Found rviz_assimp_vendor.patch
==> Validating source files with sha256sums...
    ros2 ... Skipped
    google_benchmark_vendor.patch ... Passed
    rviz_assimp_vendor.patch ... Passed
==> Making package: ros2-galactic 2021.07.16-5 (Thu 06 Jan 2022 02:20:18 AM EST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating ros2 git repo...
Fetching origin
  -> Found google_benchmark_vendor.patch
  -> Found rviz_assimp_vendor.patch
==> Validating source files with sha256sums...
    ros2 ... Skipped
    google_benchmark_vendor.patch ... Passed
    rviz_assimp_vendor.patch ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of ros2 git repo...
Cloning into 'ros2'...
done.
Switched to a new branch 'makepkg'
==> Starting prepare()...
LANG=en_US.UTF-8
Traceback (most recent call last):
  File "/usr/bin/vcs", line 33, in <module>
    sys.exit(load_entry_point('vcstool==0.3.0', 'console_scripts', 'vcs')())
  File "/usr/bin/vcs", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 919, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 518, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: No package metadata was found for vcstool
==> ERROR: A failure occurred in prepare().
    Aborting...
error: failed to build 'ros2-galactic-2021.07.16-5': 
error: packages failed to build: ros2-galactic-2021.07.16-5
```

KenwoodFox, 2022-01-06:
Fixed the python issue, for anybody else using arch/manjaro make sure to check ur python aur packages

Now i have a dep issue with colon-core? 

```
[11.219s] ERROR:colcon.colcon_core.entry_point:Exception loading extension 'colcon_core.task.build.ros.catkin': (colcon-core 0.6.1 (/usr/lib/python3.10/site-packages), Requirement.parse('colcon-core>=0.7.0'))
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/colcon_core/entry_point.py", line 120, in load_entry_points
    extension_type = load_entry_point(entry_point)
  File "/usr/lib/python3.10/site-packages/colcon_core/entry_point.py", line 168, in load_entry_point
    return entry_point.load()
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2449, in load
    self.require(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2472, in require
    items = working_set.resolve(reqs, env, installer, extras=self.extras)
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 777, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.VersionConflict: (colcon-core 0.6.1 (/usr/lib/python3.10/site-packages), Requirement.parse('colcon-core>=0.7.0'))

[11.219s] ERROR:colcon.colcon_core.entry_point:Exception loading extension 'colcon_core.task.build.ros.cmake': (colcon-core 0.6.1 (/usr/lib/python3.10/site-packages), Requirement.parse('colcon-core>=0.7.0'))
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/colcon_core/entry_point.py", line 120, in load_entry_points
    extension_type = load_entry_point(entry_point)
  File "/usr/lib/python3.10/site-packages/colcon_core/entry_point.py", line 168, in load_entry_point
    return entry_point.load()
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2449, in load
    self.require(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2472, in require
    items = working_set.resolve(reqs, env, installer, extras=self.extras)
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 777, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.VersionConflict: (colcon-core 0.6.1 (/usr/lib/python3.10/site-packages), Requirement.parse('colcon-core>=0.7.0'))

[11.220s] ERROR:colcon.colcon_core.entry_point:Exception loading extension 'colcon_core.task.build.ros.ament_cmake': (colcon-core 0.6.1 (/usr/lib/python3.10/site-packages), Requirement.parse('colcon-core>=0.7.0'))
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/colcon_core/entry_point.py", line 120, in load_entry_points
    extension_type = load_entry_point(entry_point)
  File "/usr/lib/python3.10/site-packages/colcon_core/entry_point.py", line 168, in load_entry_point
    return entry_point.load()
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2449, in load
    self.require(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2472, in require
    items = working_set.resolve(reqs, env, installer, extras=self.extras)
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 777, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.VersionConflict: (colcon-core 0.6.1 (/usr/lib/python3.10/site-packages), Requirement.parse('colcon-core>=0.7.0'))

[11.221s] ERROR:colcon.colcon_core.entry_point:Exception loading extension 'colcon_core.task.build.ros.ament_python': (colcon-core 0.6.1 (/usr/lib/python3.10/site-packages), Requirement.parse('colcon-core>=0.7.0'))
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/colcon_core/entry_point.py", line 120, in load_entry_points
    extension_type = load_entry_point(entry_point)
  File "/usr/lib/python3.10/site-packages/colcon_core/entry_point.py", line 168, in load_entry_point
    return entry_point.load()
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2449, in load
    self.require(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2472, in require
    items = working_set.resolve(reqs, env, installer, extras=self.extras)
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 777, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.VersionConflict: (colcon-core 0.6.1 (/usr/lib/python3.10/site-packages), Requirement.parse('colcon-core>=0.7.0'))

[11.221s] ERROR:colcon.colcon_core.entry_point:Exception loading extension 'colcon_core.task.build.ros.catkin': (colcon-core 0.6.1 (/usr/lib/python3.10/site-packages), Requirement.parse('colcon-core>=0.7.0'))
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/colcon_core/entry_point.py", line 120, in load_entry_points
    extension_type = load_entry_point(entry_point)
  File "/usr/lib/python3.10/site-packages/colcon_core/entry_point.py", line 168, in load_entry_point
    return entry_point.load()
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2449, in load
    self.require(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2472, in require
[... 837 lines cut ...]
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2472, in require
    items = working_set.resolve(reqs, env, installer, extras=self.extras)
[... cut ...]
    items = working_set.resolve(reqs, env, installer, extras=self.extras)
  File "/usr/lib/python3.10/site-packages/pkg_resources/__init__.py", line 777, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.VersionConflict: (colcon-core 0.6.1 (/usr/lib/python3.10/site-packages), Requirement.parse('colcon-core>=0.7.0'))

==> ERROR: A failure occurred in build().
    Aborting...
error: failed to build 'ros2-galactic-2021.07.16-5': 
error: packages failed to build: ros2-galactic-2021.07.16-5
```

KenwoodFox, 2022-01-06:
So far, ive gone through with this:

```
$ pip3 install colcon-core==0.7.0
$ pip3 install pygments
$ pip3 install markupsafe
```

and after all that, i got it to this stage

```
Finished <<< uncrustify_vendor [35.6s]
Starting >>> ament_uncrustify
Finished <<< orocos_kdl [27.0s]                                                                                                                              
Finished <<< ament_uncrustify [1.27s]                                                                                                                         
Starting >>> ament_cmake_uncrustify
--- stderr: rviz_assimp_vendor                                                                                                                                
c++: internal compiler error: Bus error signal terminated program cc1plus
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.archlinux.org/> for instructions.
make[5]: *** [code/CMakeFiles/assimp.dir/build.make:1434: code/CMakeFiles/assimp.dir/ObjExporter.cpp.o] Error 4
make[5]: *** Waiting for unfinished jobs....
make[4]: *** [CMakeFiles/Makefile2:184: code/CMakeFiles/assimp.dir/all] Error 2
make[3]: *** [Makefile:136: all] Error 2
make[2]: *** [CMakeFiles/assimp-4.1.0.dir/build.make:86: assimp-4.1.0-prefix/src/assimp-4.1.0-stamp/assimp-4.1.0-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/assimp-4.1.0.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< rviz_assimp_vendor [29.7s, exited with code 2]
Aborted  <<< iceoryx_posh [38.5s]
Aborted  <<< ament_cmake_uncrustify [0.31s]
Aborted  <<< foonathan_memory_vendor [30.1s]                                                                                    
Aborted  <<< rviz_ogre_vendor [2min 59s]                                            

Summary: 83 packages finished [3min 15s]
  1 package failed: rviz_assimp_vendor
  4 packages aborted: ament_cmake_uncrustify foonathan_memory_vendor iceoryx_posh rviz_ogre_vendor
  10 packages had stderr output: ament_cmake_google_benchmark ament_cmake_test foonathan_memory_vendor google_benchmark_vendor iceoryx_posh rviz_assimp_vendor tracetools_read tracetools_trace uncrustify_vendor urdfdom_headers
  242 packages not processed
==> ERROR: A failure occurred in build().
    Aborting...
error: failed to build 'ros2-galactic-2021.07.16-5': 
error: packages failed to build: ros2-galactic-2021.07.16-5
```

so i think im making good progress

KenwoodFox, 2022-01-06:
Im here now

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
library_abs
    linked by target "rviz_rendering" in directory /home/joe/.cache/paru/clone/ros2-galactic/src/ros2/src/ros2/rviz/rviz_rendering
```

I found this is linked to this issue:
https://github.com/ros2/ros2/issues/1222

Same failure, but i think i already see these patches applied?

m2-farzan, 2022-01-06:
Good progress @KenwoodFox. It's nice that you share the details and gotchas along the way.

Yes, the problem with rviz_rendering was already reported (https://github.com/m2-farzan/ros2-galactic-PKGBUILD/issues/10) and should have been fixed by the patch. It's strange that you see the patch applied but still get the error... No explanation comes to my mind... Maybe cleaning the build files and trying again could help?

m2-farzan, 2022-01-06:
> Its good to hear its fixed now! very interesting though.. guess ive got some fixing to do, will this happen again if i move from glactic to what comes next?..

No, we should be fine for future versions because the symlinks are now owned by a package (ros2-pyqt5-sip-compat), instead of being dangling artifacts of package preparation script (as it was the case in Foxy).

KenwoodFox, 2022-01-07:
> Good progress @KenwoodFox. It's nice that you share the details and gotchas along the way.
>
> Yes, the problem with rviz_rendering was already reported (#10) and should have been fixed by the patch. [...]

It is surely weird, heres how i see it from my end, i tried clearing the cache but, no such luck, maybe its a problem with paru, ill try with a manual pkgbuild

```
cat /home/joe/.cache/paru/clone/ros2-galactic/src/ros2/src/ros2/rviz/rviz_assimp_vendor/rviz_assimp_vendor-extras.cmake.in
# Override ON so that the following CMake logic in assimp 5.0.1 and older
# doesn't result in a CMake warning: if(ON)
#   https://github.com/ros2/rviz/issues/524
#   https://bugs.launchpad.net/ubuntu/+source/assimp/+bug/1869405
set(ON 1)

find_package(assimp QUIET)

if(NOT assimp_FOUND OR "${assimp_VERSION}" VERSION_LESS 4.1.0)
  # add the local Modules directory to the modules path
  set(assimp_DIR "${@PROJECT_NAME@_DIR}/../../../opt/rviz_assimp_vendor/lib/cmake/assimp-4.1")
  message(STATUS "Setting assimp_DIR to: '${assimp_DIR}'")
endif()

find_package(assimp REQUIRED QUIET)

# workaround bug in Assimp 4.1.0 https://github.com/assimp/assimp/issues/1914
# Affecting Ubuntu package: libassimp-dev 4.1.0~dfsg-3, Brew: assimp 4.1.0
string(REPLACE "/lib/lib/" "/lib/" ASSIMP_LIBRARY_DIRS "${ASSIMP_LIBRARY_DIRS}")
string(REGEX REPLACE "/lib/include$" "/include" ASSIMP_INCLUDE_DIRS "${ASSIMP_INCLUDE_DIRS}")

set(rviz_assimp_vendor_LIBRARIES)
foreach(library IN LISTS ASSIMP_LIBRARIES)
  message(STATUS "library: ${library}")
  if(IS_ABSOLUTE "${library}")
    list(APPEND rviz_assimp_vendor_LIBRARIES "${library}")
  else()
    find_library(library_abs assimp)
    list(APPEND rviz_assimp_vendor_LIBRARIES "${library_abs}")
  endif()
endforeach()
set(rviz_assimp_vendor_LIBRARY_DIRS ${ASSIMP_LIBRARY_DIRS})
set(rviz_assimp_vendor_INCLUDE_DIRS ${ASSIMP_INCLUDE_DIRS})
```

KenwoodFox, 2022-01-07:
Even a fresh build with `makepkg -si` is a no go, it does rebuild absolutly everything but stull does not get past `linked by target "rviz_rendering" in directory /tmp/ros2-galactic/src/ros2/src/ros2/rviz/rviz_rendering`

Luckily, i think everything other than rviz works

sad that rviz is literally the only tool that i want XD

KenwoodFox, 2022-01-07:
The dep you added in 050c750 fixed it! Thank you!

closed 2022-01-07 by KenwoodFox

m2-farzan, 2022-01-08:
Thanks for reporting the problem.

KenwoodFox, 2022-01-08:
> Thanks for reporting the problem.

Thank you for being supportive and helping me out!