m2-farzan/ros2-galactic-PKGBUILD #15: ERROR:colcon:colcon build: Some selected packages are already built in one or more underlay workspaces:

KenwoodFox, 2022-01-10:
Looks like its aborting build because ros is installed already? it seems tedious to have to uninstall then reinstall completly on every update, can overwriting be enabled?

```
[2.682s] ERROR:colcon:colcon build: Some selected packages are already built in one or more underlay workspaces:
        'ament_cppcheck' is in: /opt/ros2/galactic
        'ament_lint' is in: /opt/ros2/galactic
        'ament_package' is in: /opt/ros2/galactic
        'ament_pycodestyle' is in: /opt/ros2/galactic
        'osrf_pycommon' is in: /opt/ros2/galactic
        'ament_cmake_core' is in: /opt/ros2/galactic
        'ament_flake8' is in: /opt/ros2/galactic
        'ament_pyflakes' is in: /opt/ros2/galactic
        'ament_cmake_export_definitions' is in: /opt/ros2/galactic
        'ament_cmake_export_include_directories' is in: /opt/ros2/galactic
        'ament_cmake_export_libraries' is in: /opt/ros2/galactic
        'ament_cmake_export_link_flags' is in: /opt/ros2/galactic
        'ament_cmake_include_directories' is in: /opt/ros2/galactic
        'ament_cmake_libraries' is in: /opt/ros2/galactic
        'ament_cmake_python' is in: /opt/ros2/galactic
        'ament_cmake_version' is in: /opt/ros2/galactic
        'ament_mypy' is in: /opt/ros2/galactic
        'ament_pep257' is in: /opt/ros2/galactic
        'ros_environment' is in: /opt/ros2/galactic
        'test_interface_files' is in: /opt/ros2/galactic
        'ament_cmake_export_dependencies' is in: /opt/ros2/galactic
        'ament_cmake_export_interfaces' is in: /opt/ros2/galactic
        'ament_cmake_export_targets' is in: /opt/ros2/galactic
        'ament_cmake_target_dependencies' is in: /opt/ros2/galactic
        'ament_cmake_test' is in: /opt/ros2/galactic
        'ament_copyright' is in: /opt/ros2/galactic
        'ament_clang_format' is in: /opt/ros2/galactic
        'ament_clang_tidy' is in: /opt/ros2/galactic
        'ament_cmake' is in: /opt/ros2/galactic
        'ament_cmake_google_benchmark' is in: /opt/ros2/galactic
        'ament_cmake_gtest' is in: /opt/ros2/galactic
        'ament_cmake_nose' is in: /opt/ros2/galactic
        'ament_cmake_pytest' is in: /opt/ros2/galactic
        'ament_cpplint' is in: /opt/ros2/galactic
        'ament_index_python' is in: /opt/ros2/galactic
        'ament_lint_auto' is in: /opt/ros2/galactic
        'ament_lint_cmake' is in: /opt/ros2/galactic
        'ament_pclint' is in: /opt/ros2/galactic
        'ament_xmllint' is in: /opt/ros2/galactic
        'domain_coordinator' is in: /opt/ros2/galactic
        'ament_cmake_auto' is in: /opt/ros2/galactic
        'ament_cmake_gmock' is in: /opt/ros2/galactic
        'ament_cmake_lint_cmake' is in: /opt/ros2/galactic
        'launch' is in: /opt/ros2/galactic
        'libcurl_vendor' is in: /opt/ros2/galactic
        'pybind11_vendor' is in: /opt/ros2/galactic
        'rosidl_cli' is in: /opt/ros2/galactic
        'rpyutils' is in: /opt/ros2/galactic
        'shared_queues_vendor' is in: /opt/ros2/galactic
        'sqlite3_vendor' is in: /opt/ros2/galactic
        'tinyxml2_vendor' is in: /opt/ros2/galactic
        'tinyxml_vendor' is in: /opt/ros2/galactic
        'tracetools_read' is in: /opt/ros2/galactic
        'tracetools_trace' is in: /opt/ros2/galactic
        'uncrustify_vendor' is in: /opt/ros2/galactic
        'yaml_cpp_vendor' is in: /opt/ros2/galactic
        'zstd_vendor' is in: /opt/ros2/galactic
        'ament_cmake_copyright' is in: /opt/ros2/galactic
        'ament_uncrustify' is in: /opt/ros2/galactic
        'launch_testing' is in: /opt/ros2/galactic
        'launch_xml' is in: /opt/ros2/galactic
        'launch_yaml' is in: /opt/ros2/galactic
        'ament_cmake_clang_format' is in: /opt/ros2/galactic
        'ament_cmake_clang_tidy' is in: /opt/ros2/galactic
        'ament_cmake_cppcheck' is in: /opt/ros2/galactic
        'ament_cmake_cpplint' is in: /opt/ros2/galactic
[... 233 lines cut ...]
        'rosbag2_transport' is in: /opt/ros2/galactic
        'rviz_default_plugins' is in: /opt/ros2/galactic
        'sros2_cmake' is in: /opt/ros2/galactic
        'test_launch_ros' is in: /opt/ros2/galactic
        'test_security' is in: /opt/ros2/galactic
        'pendulum_control' is in: /opt/ros2/galactic
        'ros1_bridge' is in: /opt/ros2/galactic
        'ros2component' is in: /opt/ros2/galactic
        'rosbag2_py' is in: /opt/ros2/galactic
        'rviz2' is in: /opt/ros2/galactic
        'ros2bag' is in: /opt/ros2/galactic
        'ros2cli_common_extensions' is in: /opt/ros2/galactic
        'rqt_bag' is in: /opt/ros2/galactic
        'rosbag2_tests' is in: /opt/ros2/galactic
        'rosbag2' is in: /opt/ros2/galactic
        'rqt_bag_plugins' is in: /opt/ros2/galactic
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
        --allow-overriding action_msgs action_tutorials_cpp action_tutorials_interfaces action_tutorials_py actionlib_msgs ament_clang_format ament_clang_tidy ament_cmake ament_cmake_auto ament_cmake_clang_format ament_cmake_clang_tidy ament_cmake_copyright ament_cmake_core ament_cmake_cppcheck ament_cmake_cpplint ament_cmake_export_definitions ament_cmake_export_dependencies ament_cmake_export_include_directories ament_cmake_export_interfaces ament_cmake_export_libraries ament_cmake_export_link_flags ament_cmake_export_targets ament_cmake_flake8 ament_cmake_gmock ament_cmake_google_benchmark ament_cmake_gtest ament_cmake_include_directories ament_cmake_libraries ament_cmake_lint_cmake ament_cmake_mypy ament_cmake_nose ament_cmake_pclint ament_cmake_pep257 ament_cmake_pycodestyle ament_cmake_pyflakes ament_cmake_pytest ament_cmake_python ament_cmake_ros ament_cmake_target_dependencies ament_cmake_test ament_cmake_uncrustify ament_cmake_version ament_cmake_xmllint ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_cpp ament_index_python ament_lint ament_lint_auto ament_lint_cmake ament_lint_common ament_mypy ament_package ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_uncrustify ament_xmllint builtin_interfaces camera_calibration_parsers camera_info_manager class_loader common_interfaces composition composition_interfaces console_bridge_vendor demo_nodes_cpp demo_nodes_cpp_native demo_nodes_py diagnostic_msgs domain_coordinator dummy_map_server dummy_robot_bringup dummy_sensors eigen3_cmake_module example_interfaces examples_rclcpp_cbg_executor examples_rclcpp_minimal_action_client examples_rclcpp_minimal_action_server examples_rclcpp_minimal_client examples_rclcpp_minimal_composition examples_rclcpp_minimal_publisher examples_rclcpp_minimal_service examples_rclcpp_minimal_subscriber examples_rclcpp_minimal_timer examples_rclcpp_multithreaded_executor examples_rclpy_executors examples_rclpy_guard_conditions examples_rclpy_minimal_action_client examples_rclpy_minimal_action_server examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber examples_rclpy_pointcloud_publisher examples_tf2_py fastrtps_cmake_module geometry2 geometry_msgs image_common image_tools image_transport interactive_markers intra_process_demo kdl_parser laser_geometry launch launch_ros launch_testing launch_testing_ament_cmake launch_testing_ros launch_xml launch_yaml libcurl_vendor libstatistics_collector libyaml_vendor lifecycle lifecycle_msgs logging_demo map_msgs message_filters mimick_vendor nav_msgs osrf_pycommon pendulum_control pendulum_msgs performance_test_fixture pluginlib pybind11_vendor python_cmake_module python_qt_binding qt_dotgraph qt_gui qt_gui_app qt_gui_core qt_gui_cpp qt_gui_py_common quality_of_service_demo_cpp quality_of_service_demo_py rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_interface rcl_logging_log4cxx rcl_logging_noop rcl_logging_spdlog rcl_yaml_param_parser rclcpp rclcpp_action rclcpp_components rclcpp_lifecycle rclpy rcpputils rcutils resource_retriever rmw rmw_connextdds rmw_connextdds_common rmw_connextddsmicro rmw_cyclonedds_cpp rmw_dds_common rmw_fastrtps_cpp [... cut ...] l tf2_geometry_msgs tf2_kdl tf2_msgs tf2_py tf2_ros tf2_ros_py tf2_sensor_msgs tf2_tools tinyxml2_vendor tinyxml_vendor tlsf tlsf_cpp topic_monitor topic_statistics_demo tracetools tracetools_launch tracetools_read tracetools_test tracetools_trace trajectory_msgs turtlesim uncrustify_vendor unique_identifier_msgs urdf urdf_parser_plugin visualization_msgs yaml_cpp_vendor zstd_vendor
==> ERROR: A failure occurred in build().
    Aborting...
error: failed to build 'ros2-galactic-2021.07.16-6':
```

KenwoodFox, 2022-01-10:
> Does the error show up even if you don't run `source /opt/ros2/galactic/setup.bash` in the terminal? (The source command might be implicit in ~/.bashrc or ~/.bash_profile)

Ooh, good question, i can experiment, uninstalling `ros2` worked fine, i was able to update no problem after that but it was a tad annoying, i can see if removing `/opt/ros2/galactic/setup.zsh` from my .zshrc also fixes this

m2-farzan, 2022-06-07:
@KenwoodFox I'll close this for now. Feel free to re-open it if needed.

closed 2022-06-07 by m2-farzan