m2-farzan/ros2-galactic-PKGBUILD #18: Have the package build source the latest tagged ros2-galactic release

andrewstockwell, 2022-01-30:
The github url seems to be set on a static tag. It's possible to have the package pull from the latest released tag, or otherwise using sed. [This wiki article](https://aur.archlinux.org/ros2-galactic.git) seems to provide a decent way to do this. I'm not sure what implications this has for marking the package as updated when a new tag comes out.

m2-farzan, 2022-02-03:
Thanks for the suggestion but new releases will likely break the package. I think we should keep the manual step for the sake of stability. Besides, we have ros2-git package which is not meant to be super stable but it always uses latest rolling version.

Also, galactic currently points to the latest release, and whenever a new galactic release comes, I'll update the package, test it and push it to AUR. If I don't see a new update, users can flag the package outdated and I'll be notified by email.

P.S. I think the link in your post is different from what it was intended to be.

andrewstockwell, 2022-02-03:
Thanks for the explanation! I think I'll close this given that info.

closed 2022-02-03 by andrewstockwell