m2-farzan/ros2-galactic-PKGBUILD #19: Build error: package failed: `tlsf_cpp`

zjeffer, 2022-02-23:
Here's the output after running `paru -S ros2-galactic`:

[output.txt](https://github.com/m2-farzan/ros2-galactic-PKGBUILD/files/8124732/output.txt)

This is with Python 3.10. Is my Python version too new?

zjeffer, 2022-02-23:
I fixed the issue by symlinking Python 3.9 instead of 3.10 like this:

```sh
sudo ln -sf /usr/bin/python3.9 /usr/bin/python3
```

When installing ros2-galactic, I came across some other errors that indicated missing packages. Simply installing these packages with Python3.9's pip fixed the installation.

As Arch Linux now uses Python 3.10 by default, a better fix would be to make it work with that version.

m2-farzan, 2022-02-23:
Thanks for reporting the issue. It's reproducible.

It's an strange one because you have solved it by changing python version (but from the logs it seems to be a C++ error)... How did you figure out the fact that the error is related to python version? (I'm asking this because it would be a starting point for me to find the cause of the issue and hopefully fix it)...

zjeffer, 2022-02-23:
A friend of mine who installed the same package today uses Python 3.9, and not 3.10. He didn't run into the issue.

m2-farzan, 2022-02-24:
I've modified the package to use an upstream fix. Waiting for the test: https://github.com/m2-farzan/ros2-galactic-PKGBUILD/runs/5324847685?check_suite_focus=true

zjeffer, 2022-02-24:
The new package builds without the tlsf_cpp errors on Python 3.10 (but still only 0.02MiB like in #20 ).

I guess I should wait until the test completes successfully before closing this issue?

lockeadams, 2022-02-25:
After retrying with the new package on Python 3.10.2 it successfully got past the tlsf_cpp errors, but it failed with a very similar issue while building pendulum_control. Is there a similar fix for that one?

Here's some output from the error:
```
Starting >>> pendulum_control    
--- stderr: pendulum_control                                           
/home/locke/.cache/yay/ros2-galactic/src/ros2/src/ros2/demos/pendulum_control/src/pendulum_demo.cpp: In function ‘void* testing_malloc(size_t, const void*)’:
/home/locke/.cache/yay/ros2-galactic/src/ros2/src/ros2/demos/pendulum_control/src/pendulum_demo.cpp:63:3: error: ‘__malloc_hook’ was not declared in this scope; did you mean ‘prev_malloc_hook’?
   63 |   __malloc_hook = prev_malloc_hook;
      |   ^~~~~~~~~~~~~
      |   prev_malloc_hook
/home/locke/.cache/yay/ros2-galactic/src/ros2/src/ros2/demos/pendulum_control/src/pendulum_demo.cpp: In function ‘void init_malloc_hook()’:
/home/locke/.cache/yay/ros2-galactic/src/ros2/src/ros2/demos/pendulum_control/src/pendulum_demo.cpp:82:22: error: ‘__malloc_hook’ was not declared in this scope; did you mean ‘prev_malloc_hook’?
   82 |   prev_malloc_hook = __malloc_hook;
      |                      ^~~~~~~~~~~~~
      |                      prev_malloc_hook
make[2]: *** [CMakeFiles/pendulum_demo.dir/build.make:76: CMakeFiles/pendulum_demo.dir/src/pendulum_demo.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/pendulum_demo.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:146: all] Error 2
---
Failed   <<< pendulum_control [13.2s, exited with code 2]
```

MightyPiggie, 2022-02-25:
> After retrying with the new package on Python 3.10.2 it successfully got past the tlsf_cpp errors, but it failed with a very similar issue while building pendulum_control. Is there a similar fix for [...]

I can confirm this problem. I am also running python 3.10.2.

m2-farzan, 2022-02-25:
Pushed a similar fix for pendulum_control. Waiting for the test: https://github.com/m2-farzan/ros2-galactic-PKGBUILD/runs/5332494593?check_suite_focus=true

m2-farzan, 2022-02-25:
The tests are passing!

Thanks everybody for reporting the issues. Don't forget to star the repo if you like :)

closed 2022-02-25 by m2-farzan