m2-farzan/ros2-galactic-PKGBUILD #21: RQT does not have the image visualizer plugin

zjeffer, 2022-02-28:
I have a /video topic, but when I open RQT using `ros2 run rqt_gui rqt_gui`, and go to `Plugins > Visualisation`, there is no option to show images:

[image]

The option is there when using the docker image `ros:galactic`, but not when installing from this package. What am I doing wrong?

m2-farzan, 2022-02-28:
I don't have much experience with rqt but you may need to install a plugin (maybe [this one](http://wiki.ros.org/rqt_image_view)?)

Alternatively try rviz for visualizing image topics. Run `rviz2` in terminal.

m2-farzan, 2022-02-28:
I too don't have experience with rqt plugins. I'm afraid I can't help with this one.

goekce, 2022-03-05:
As Mostafa mentioned `rqt_image_view` should help.

In my case installing and compiling `rqt_common_plugins` and `image_transport_plugins` using rosinstall_generator helped. These  are included in the [PKGBUILD of mine for Turtlebot3 development](https://mygit.th-deg.de/gaydos/pkgbuilds/-/blob/main/ros-galactic-turtlebot3-quickstart/PKGBUILD#L9).

Much easier approach is to compile them in a workspace and activate the workspace. They should be then visible in the rqt.

closed 2022-03-08 by zjeffer