m2-farzan/ros2-galactic-PKGBUILD #22: Failing to build because global git user mail / name are not set

schlagenhauf, 2022-03-01:
Thanks for maintaining this package!

Trying to build / install ros2 with `aur sync ros2-galactic` fails with

```
HEAD is now at a76814d 2.5.2
=== /home/pntagrl/.cache/aurutils/sync/ros2-galactic/src/ros2/src/ros2/yaml_cpp_vendor (git) ===

Previous HEAD position was a00f059 Fix handling of CMAKE_C[XX]_FLAGS lists (#24)
HEAD is now at 03a4cd7 7.1.0
Updated 3 paths from the index
Previous HEAD position was 2921d4402 Merge pull request #735 from ApexAI/iox-#482-three-pillars
HEAD is now at a07d83b7c Merge pull request #836 from ApexAI/iox-#828-iceoryx-release-1.0.1
Previous HEAD position was 9db463c8 Advertise addresses independent of ManySocketsMode
HEAD is now at c59f7522 Rename libidl to libcycloneddsidl
Previous HEAD position was 03a4cd7 7.1.0
HEAD is now at a00f059 Fix handling of CMAKE_C[XX]_FLAGS lists (#24)
Committer identity unknown

*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.

fatal: unable to auto-detect email address (got 'pntagrl@Nebucharch.(none)')
==> ERROR: A failure occurred in prepare().
    Aborting...
```

Setting global mail / name lets me continue but it would be nice if there would be a way around setting / unsetting it for every update.

goekce, 2022-03-06:
I also encountered the same problem while building the package in a clean chroot. My workaround was to give a domain to the machine (in my case in chroot).

`/etc/hosts`:
```
127.0.0.1 hostname.localdomain hostname
```

m2-farzan, 2022-03-07:
Thanks @schlagenhauf for reporting the issue and @goekce for sharing your workaround. I've pushed a patch based on environment variables. The issue should be fixed now.

closed 2022-03-07 by m2-farzan