m2-farzan/ros2-galactic-PKGBUILD #23: Build error `error: ‘memcpy’ was not declared in this scope`

kistlin, 2022-06-06:
Tried to install `ros2-galactic 2021.07.16-12` from AUR with `paru`.

Error output
```
Finished <<< ament_copyright [7.62s]
Starting >>> ament_index_python
Starting >>> domain_coordinator
Starting >>> ament_xmllint
Starting >>> ament_lint_cmake
Starting >>> ament_cpplint
Starting >>> ament_clang_format
Starting >>> ament_clang_tidy
Finished <<< ament_cmake [6.88s]
Starting >>> uncrustify_vendor
Finished <<< ament_cmake_gmock [10.1s]
Starting >>> tinyxml2_vendor
--- stderr: osrf_testing_tools_cpp
ros2-galactic/src/ros2/src/osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/memory_tools/is_working.cpp: In function ‘void osrf_testing_tools_cpp::memory_tools::guaranteed_malloc(const std::string&)’:
ros2-galactic/src/ros2/src/osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/memory_tools/is_working.cpp:37:3: error: ‘memcpy’ was not declared in this scope
   37 |   memcpy(some_memory, str.c_str(), str.length());
      |   ^~~~~~
ros2-galactic/src/ros2/src/osrf/osrf_testing_tools_cpp/osrf_testing_tools_cpp/src/memory_tools/is_working.cpp:20:1: note: ‘memcpy’ is defined in header ‘<cstring>’; did you forget to ‘#include <cstring>’?
   19 | #include "osrf_testing_tools_cpp/memory_tools/register_hooks.hpp"
  +++ |+#include <cstring>
   20 |
make[2]: *** [src/memory_tools/CMakeFiles/memory_tools.dir/build.make:118: src/memory_tools/CMakeFiles/memory_tools.dir/is_working.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1012: src/memory_tools/CMakeFiles/memory_tools.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:146: all] Error 2
---
Failed   <<< osrf_testing_tools_cpp [35.2s, exited with code 2]
```

m2-farzan, 2022-06-06:
Thanks for reporting the issue. It should be fixed now.

Testing now: https://github.com/m2-farzan/ros2-galactic-PKGBUILD/runs/6754202190

kistlin, 2022-06-06:
Thanks @m2-farzan. It got significantly further. I still get an error at
```
Finished <<< rti_connext_dds_cmake_module [6.49s]
Starting >>> tango_icons_vendor
Finished <<< spdlog_vendor [6.59s]                                                                                                                                                                                                                                                                                                                         
Starting >>> performance_test_fixture
Finished <<< launch_testing_ament_cmake [5.66s]                                                                                                                                                                                                                                                                                                                                     
Starting >>> urdf_parser_plugin
Finished <<< rosidl_adapter [11.9s]                                                                                                                                                                                                                                                                                                                                     
Starting >>> rosidl_parser                                                                                                                                                                                                                                                                                                                                                 
--- stderr: performance_test_fixture                                                                                                                                                                                                                                                                                                                                         
ros2-galactic/src/ros2/src/ros2/performance_test_fixture/src/performance_test_fixture.cpp: In constructor ‘performance_test_fixture::PerformanceTest::PerformanceTest()’:
ros2-galactic/src/ros2/src/ros2/performance_test_fixture/src/performance_test_fixture.cpp:34:5: error: ‘strcmp’ was not declared in this scope
   34 |     strcmp("1", performance_test_fixture_enable_trace) == 0)
      |     ^~~~~~
ros2-galactic/src/ros2/src/ros2/performance_test_fixture/src/performance_test_fixture.cpp:19:1: note: ‘strcmp’ is defined in header ‘<cstring>’; did you forget to ‘#include <cstring>’?
   18 | #include <cstdlib>
  +++ |+#include <cstring>
   19 | #include <vector>
make[2]: *** [CMakeFiles/performance_test_fixture.dir/build.make:76: CMakeFiles/performance_test_fixture.dir/src/performance_test_fixture.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:143: CMakeFiles/performance_test_fixture.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

kistlin, 2022-06-07:
Thanks for the quick fixes. It now got passed it, but it fails at
```
Starting >>> kdl_parser
Finished <<< kdl_parser [12.4s]                                                                                                                                                       
--- stderr: qt_gui_cpp                                                                                                                                
sip: /usr/share/sip/PyQt5/QtCore/QtCoremod.sip:23: syntax error
Traceback (most recent call last):
  File "ros2-galactic/src/install/share/python_qt_binding/cmake/sip_configure.py", line 116, in <module>
    subprocess.check_call(cmd)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/sip', '-c', 'ros2-galactic/src/build/qt_gui_cpp/sip/qt_gui_cpp_sip', '-b', 'ros2-galactic/src/build/qt_gui_cpp/sip/qt_gui_cpp_sip/pyqtscripting.sbf', '-I', '/usr/share/sip/PyQt5', '-w', '-n', 'PyQt5.sip', '-t', 'Qt_5_15_0', '-t', 'WS_X11', 'qt_gui_cpp.sip']' returned non-zero exit status 1.
make[2]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/build.make:103: sip/qt_gui_cpp_sip/Makefile] Error 1
make[1]: *** [CMakeFiles/Makefile2:284: src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

The line 23 mentioned in the error in file `/usr/share/sip/PyQt5/QtCore/QtCoremod.sip`.
```
...
%Module(name=PyQt5.QtCore, call_super_init=True, default_VirtualErrorHandler=PyQt5, keyword_arguments="Optional", use_limited_api=True, py_ssize_t_clean=True)
```
The timeline does not contain the minor version of my current Qt5 version. Not sure if relevant.
```
%Timeline {Qt_5_0_0 Qt_5_0_1 Qt_5_0_2 Qt_5_1_0 Qt_5_1_1 Qt_5_2_0 Qt_5_2_1 Qt_5_3_0 Qt_5_3_1 Qt_5_3_2 Qt_5_4_0 Qt_5_4_1 Qt_5_4_2 Qt_5_5_0 Qt_5_5_1 Qt_5_6_0 Qt_5_6_1 Qt_5_6_2 Qt_5_6_3 Qt_5_6_4 Qt_5_6_5 Qt_5_6_6 Qt_5_6_7 Qt_5_6_8 Qt_5_6_9 Qt_5_7_0 Qt_5_7_1 Qt_5_8_0 Qt_5_8_1 Qt_5_9_0 Qt_5_9_1 Qt_5_9_2 Qt_5_9_3 Qt_5_9_4 Qt_5_9_5 Qt_5_9_6 Qt_5_9_7 Qt_5_9_8 Qt_5_9_9 Qt_5_10_0 Qt_5_10_1 Qt_5_11_0 Qt_5_11_1 Qt_5_11_2 Qt_5_11_3 Qt_5_12_0 Qt_5_12_1 Qt_5_12_2 Qt_5_12_3 Qt_5_12_4 Qt_5_13_0 Qt_5_14_0 Qt_5_15_0}
...
```

The folder `ros2-galactic/src/build/qt_gui_cpp/sip/qt_gui_cpp_sip` is empty. Not sure if that is good. `sip` does not have the best help output... .

If I run 
```
/usr/bin/sip -c ros2-galactic/src/build/qt_gui_cpp/sip/qt_gui_cpp_sip -b ros2-galactic/src/build/qt_gui_cpp/sip/qt_gui_cpp_sip/pyqtscripting.sbf -I /usr/share/sip/PyQt5 -w -n PyQt5.sip -t Qt_5_15_0 -t WS_X11 qt_gui_cpp.sip
```
 in the command line, I get the output
```
sip: Unable to find file "qt_gui_cpp.sip"
```

Installed versions
- `extra/qt5-base 5.15.4+kde+r146-1 (qt qt5) [installed]`
- `extra/python-pyqt5 5.15.6-7 (pyqt5) [installed]`
- `extra/python-pyqt5-sip 12.10.1-1 [installed]`

m2-farzan, 2022-06-07:
This is a known issue with pyqt v5.15.6 and later. Please install pyqt 5.15.5 using `sudo pip3 install pyqt5==5.15.5`.

More on this at #9.

kistlin, 2022-06-07:
Alright thanks for the fixes and guidance. It now compiled and installed successfully. I did not yet run much.
If I run `/opt/ros2/galactic/lib/turtlesim/turtlesim_node` a window with turtle pops up :).

closed 2022-06-07 by kistlin

m2-farzan, 2022-06-07:
Cheers!

Building major ROS 2 packages on Arch can be challenging. Make sure to check out [this repo](https://github.com/m2-farzan/ros2-supplementary-pkgbuilds) for some clues on how to patch common packages like nav2 for building on Arch. The instructions may be out of date as I've stopped working on that repo, but I'm open to PR's. Just in case...