m2-farzan/ros2-galactic-PKGBUILD #25: failed to install ros2-arch-deps

yummy527, 2022-06-24:
```bash
wangxl@archlinux ~/Downloads % cd ros2-arch-deps                                                                                                                                                                                                                                                                        [0]
wangxl@archlinux ~/Downloads/ros2-arch-deps % makepkg -si                                                                                                                                                                                                                                                               [0]
==> Making package: ros2-arch-deps 0.10-2 (Fri Jun 24 11:24:25 2022)
==> Checking runtime dependencies...
==> Installing missing dependencies...
[sudo] password for wangxl: 
error: target not found: python-vcstool
error: target not found: python-colcon-argcomplete
error: target not found: python-colcon-bash
error: target not found: python-catkin_pkg
error: target not found: python-colcon-cmake
error: target not found: python-colcon-core
error: target not found: python-colcon-defaults
error: target not found: python-colcon-devtools
error: target not found: python-colcon-library-path
error: target not found: python-colcon-metadata
error: target not found: python-colcon-notification
error: target not found: python-colcon-output
error: target not found: python-colcon-package-information
error: target not found: python-colcon-package-selection
error: target not found: python-colcon-pkg-config
error: target not found: python-colcon-parallel-executor
error: target not found: python-colcon-powershell
error: target not found: python-colcon-python-setup-py
error: target not found: python-colcon-recursive-crawl
error: target not found: python-colcon-ros
error: target not found: python-colcon-test-result
error: target not found: python-colcon-zsh
error: target not found: python-colcon-common-extensions
error: target not found: python-rospkg
error: target not found: python-rosdistro
error: target not found: python-rosdep
==> ERROR: 'pacman' failed to install missing dependencies.
==> Missing dependencies:
  -> python-vcstool
  -> python-colcon-argcomplete
  -> python-colcon-bash
  -> python-catkin_pkg
  -> python-colcon-cmake
  -> python-colcon-core
  -> python-colcon-defaults
  -> python-colcon-devtools
  -> python-colcon-library-path
  -> python-colcon-metadata
  -> python-colcon-notification
  -> python-colcon-output
  -> python-colcon-package-information
  -> python-colcon-package-selection
  -> python-colcon-pkg-config
  -> python-colcon-parallel-executor
  -> python-colcon-powershell
  -> python-colcon-python-setup-py
  -> python-colcon-recursive-crawl
  -> python-colcon-ros
  -> python-colcon-test-result
  -> python-colcon-zsh
  -> python-colcon-common-extensions
  -> python-rospkg
  -> python-rosdistro
  -> python-rosdep
==> Checking buildtime dependencies...
==> ERROR: Could not resolve all dependencies.

wangxl@archlinux ~/Downloads/python-vcstool % makepkg -si                                                                                                                                                                                                                                                               [0]
==> WARNING: A package has already been built, installing existing package...
==> Installing package python-vcstool with pacman -U...
loading packages...
resolving dependencies...
looking for conflicting packages...
[... 69 lines cut ...]
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/commands/command.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/commands/custom.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/commands/diff.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/commands/export.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/commands/help.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/commands/import_.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/commands/log.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/commands/pull.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/commands/push.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/commands/remotes.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/commands/status.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/commands/validate.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/commands/vcs.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/crawler.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/executor.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/streams.py exists in filesystem
python-vcstool: /opt/anaconda/lib/python3.8/site-packages/vcstool/util.py exists in filesystem
python-vcstool: /opt/anaconda/share/vcstool-completion/vcs.bash exists in filesystem
python-vcstool: /opt/anaconda/share/vcstool-completion/vcs.fish exists in filesystem
python-vcstool: /opt/anaconda/share/vcstool-completion/vcs.tcsh exists in filesystem
python-vcstool: /opt/anaconda/share/vcstool-completion/vcs.zsh exists in filesystem
Errors occurred, no packages were upgraded.
==> WARNING: Failed to install built package(s).

```

m2-farzan, 2022-06-25:
Couple of points:

- I recommend using an AUR helper like yay, over manual makepkg's. AUR helpers come with some gotchas that makes them not suitable for beginner arch users, but for big packages like ros2-galactic they can help with cumbersome dependency resolutions. In practice you'll often hop between yay and makepkg, depending on the size of the package to be installed, etc.
- It looks like you're sourcing an anaconda environment (probably in your .bashrc file) when building python3-* packages. This should be avoided.

yummy527, 2022-06-25:
> 

Yes, yay helps this.

closed 2022-06-26 by m2-farzan