m2-farzan/ros2-galactic-PKGBUILD #26: qt_gui_cpp erro with ros2-galactic-PKGBUILD

yummy527, 2022-06-25:
```shell
--- stderr: qt_gui_cpp                                                                                                                                
sip: /usr/share/sip/PyQt5/QtCore/QtCoremod.sip:23: syntax error
Traceback (most recent call last):
  File "/home/wangxl/Downloads/ros2-galactic/src/install/share/python_qt_binding/cmake/sip_configure.py", line 116, in <module>
    subprocess.check_call(cmd)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/sip', '-c', '/home/wangxl/Downloads/ros2-galactic/src/build/qt_gui_cpp/sip/qt_gui_cpp_sip', '-b', '/home/wangxl/Downloads/ros2-galactic/src/build/qt_gui_cpp/sip/qt_gui_cpp_sip/pyqtscripting.sbf', '-I', '/usr/share/sip/PyQt5', '-w', '-n', 'PyQt5.sip', '-t', 'Qt_5_15_0', '-t', 'WS_X11', 'qt_gui_cpp.sip']' returned non-zero exit status 1.
make[2]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/build.make:103: sip/qt_gui_cpp_sip/Makefile] Error 1
make[1]: *** [CMakeFiles/Makefile2:284: src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< qt_gui_cpp [25.9s, exited with code 2]
Aborted  <<< fastrtps [2min 33s]      
```

The error mentioned by "sip: /usr/share/sip/PyQt5/QtCore/QtCoremod.sip:23: syntax error" is as below:
```c
%Module(name=PyQt5.QtCore, call_super_init=True, default_VirtualErrorHandler=PyQt5, keyword_arguments="Optional", use_limited_api=True, py_ssize_t_clean=True)
```

I have tried to install ros2 on my arch linux machine for about two days. Too many warnings and errors, without even one success.
Help me!

m2-farzan, 2022-06-25:
See https://github.com/m2-farzan/ros2-galactic-PKGBUILD/issues/9.

Sorry for the inconvenience. ROS 2 on arch can be troublesome, so feel free to ask about the upcoming issues here.

yummy527, 2022-06-25:
> 

https://github.com/m2-farzan/ros2-galactic-PKGBUILD/issues/9#issuecomment-1008111252
This helps.
```shell
sudo pip3 install pyqt5==5.15.5
```

closed 2022-06-25 by yummy527