m2-farzan/ros2-galactic-PKGBUILD #27: error: ‘__malloc_hook’ was not declared in this scope

yummy527, 2022-06-25:
Help me with a new error. 
Sorry to bother you.  I googled, but found nothing helpful. And not familiar with c++.

```shell
--- stderr: tlsf_cpp                                                                                           
/home/wangxl/ros2_ws/src/ros2/realtime_support/tlsf_cpp/test/test_tlsf.cpp: In function ‘void* testing_malloc(size_t, const void*)’:
/home/wangxl/ros2_ws/src/ros2/realtime_support/tlsf_cpp/test/test_tlsf.cpp:72:3: error: ‘__malloc_hook’ was not declared in this scope; did you mean ‘prev_malloc_hook’?
   72 |   __malloc_hook = prev_malloc_hook;
      |   ^~~~~~~~~~~~~
      |   prev_malloc_hook
/home/wangxl/ros2_ws/src/ros2/realtime_support/tlsf_cpp/test/test_tlsf.cpp: In function ‘void init_malloc_hook()’:
/home/wangxl/ros2_ws/src/ros2/realtime_support/tlsf_cpp/test/test_tlsf.cpp:89:22: error: ‘__malloc_hook’ was not declared in this scope; did you mean ‘prev_malloc_hook’?
   89 |   prev_malloc_hook = __malloc_hook;
      |                      ^~~~~~~~~~~~~
      |                      prev_malloc_hook
/home/wangxl/ros2_ws/src/ros2/realtime_support/tlsf_cpp/test/test_tlsf.cpp: In function ‘void* operator new(std::size_t)’:
/home/wangxl/ros2_ws/src/ros2/realtime_support/tlsf_cpp/test/test_tlsf.cpp:216:3: error: ‘__malloc_hook’ was not declared in this scope; did you mean ‘prev_malloc_hook’?
  216 |   __malloc_hook = prev_malloc_hook;
      |   ^~~~~~~~~~~~~
      |   prev_malloc_hook
/home/wangxl/ros2_ws/src/ros2/realtime_support/tlsf_cpp/test/test_tlsf.cpp: In function ‘void operator delete(void*)’:
/home/wangxl/ros2_ws/src/ros2/realtime_support/tlsf_cpp/test/test_tlsf.cpp:233:3: error: ‘__malloc_hook’ was not declared in this scope; did you mean ‘prev_malloc_hook’?
  233 |   __malloc_hook = prev_malloc_hook;
      |   ^~~~~~~~~~~~~
      |   prev_malloc_hook
/home/wangxl/ros2_ws/src/ros2/realtime_support/tlsf_cpp/test/test_tlsf.cpp: At global scope:
/home/wangxl/ros2_ws/src/ros2/realtime_support/tlsf_cpp/test/test_tlsf.cpp:67:15: warning: ‘void* testing_malloc(size_t, const void*)’ defined but not used [-Wunused-function]
   67 | static void * testing_malloc(size_t size, const void * caller)
      |               ^~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/test_tlsf__rmw_fastrtps_dynamic_cpp.dir/build.make:76: CMakeFiles/test_tlsf__rmw_fastrtps_dynamic_cpp.dir/test/test_tlsf.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:242: CMakeFiles/test_tlsf__rmw_fastrtps_dynamic_cpp.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< tlsf_cpp [18.0s, exited with code 2]
Aborted  <<< launch_ros [4.60s]                                                         
Aborted  <<< message_filters [5.76s]                                                    
Aborted  <<< rqt_gui_cpp [19.7s]                                                        
Aborted  <<< common_interfaces [6.31s]
Aborted  <<< laser_geometry [19.8s]
Aborted  <<< ros2cli [6.32s]                                                            
Aborted  <<< examples_rclcpp_minimal_timer [7.58s]                                     
Aborted  <<< examples_rclcpp_multithreaded_executor [7.67s]                            
Aborted  <<< map_msgs [23.4s]                                                           
Aborted  <<< rclcpp_components [23.1s]                                                      
Aborted  <<< rclcpp_lifecycle [25.5s]                                                       
Aborted  <<< intra_process_demo [13.9s]                                                     
Aborted  <<< rclcpp_action [28.9s]                                                           
Aborted  <<< rosbag2_storage [17.4s]                                                           
Aborted  <<< test_quality_of_service [17.3s]                                           

Summary: 188 packages finished [3min 35s]
  1 package failed: tlsf_cpp
  15 packages aborted: common_interfaces examples_rclcpp_minimal_timer examples_rclcpp_multithreaded_executor intra_process_demo laser_geometry launch_ros map_msgs message_filters rclcpp_action rclcpp_components rclcpp_lifecycle ros2cli rosbag2_storage rqt_gui_cpp test_quality_of_service
  32 packages had stderr output: ament_clang_format ament_clang_tidy ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_mypy ament_package ament_pclint ament_pep257 ament_pep8 ament_pyflakes ament_uncrustify ament_xmllint domain_coordinator intra_process_demo launch launch_testing launch_xml launch_yaml osrf_pycommon rcl rcl_action rclcpp ros2cli rosidl_runtime_py tlsf_cpp tracetools_read tracetools_trace
  94 packages not processed

```

m2-farzan, 2022-06-25:
Thanks for reporting the issue. It is caused by the new version of glibc. I'll fix the package in a few minutes.

m2-farzan, 2022-06-25:
Oops. The issue has already been fixed.

https://github.com/m2-farzan/ros2-galactic-PKGBUILD/blob/b33f75c569c99a8a475eb892271abc30ab35802e/PKGBUILD#L77

It was first reported in https://github.com/m2-farzan/ros2-galactic-PKGBUILD/issues/19

I'm not sure why you're getting it though. Maybe the patching step has failed.

Try navigating to `/home/wangxl/ros2_ws/src/ros2/realtime_support/` and running `git log`. You should see `Remove the use of malloc hooks from the tlsf_cpp tests` in HEAD. If not, it means that the patching was unsuccessful. Try `makepkg --nobuild` in a new directrory, and investigate terminal outputs, git logs, etc. Something is failing in the `prepare` step of the pkgbuild. Maybe a simple clean-build will fix it.

yummy527, 2022-06-25:
> Oops. The issue has already been fixed.
>
> https://github.com/m2-farzan/ros2-galactic-PKGBUILD/blob/b33f75c569c99a8a475eb892271abc30ab35802e/PKGBUILD#L77
>
> It was first reported in #19
>
> I'm not sure [...]

Succeed with a clean build, finally! And I notice that issue was triggered by repos of others(may be from [here](https://wiki.archlinux.org/title/ROS) ),  not your ros2-galactic-PKGBUILD.
Thanks for all your kind work and help.

m2-farzan, 2022-06-26:
Cheers!

closed 2022-06-26 by m2-farzan