m2-farzan/ros2-galactic-PKGBUILD #30: disabling parallel build makes building process painfully slow

AnthonyMichaelTDM, 2022-08-04:
16gb is way more than enough to build the package stably. I've done it with as little as 3gb in a VM with no issues (building ros2 galactic from source, not w/ this package), even when setting the --parallel-work flag to 10 (which builds the source in ~30 minutes on my machine compared to ~4hours w/o parallel build)

maybe I'm missing something but I think the ram requirement can be safely lowered to a much lower value w/o issue

perhaps basing it on available CPU cores would be better?

m2-farzan, 2022-08-04:
Thanks for your suggestion and supporting information. The 16GB criteria was originally introduced following a [report](https://github.com/m2-farzan/ros2-foxy-PKGBUILD/pull/6) on an 8GB machine running out of memory during the build process. So I'm not too sure whether or not lowering the bar is safe. Let's keep this issue open in the hope of receiving a few more reports. Until then, I'm gonna err on the side of stability.

wojciechmadry, 2022-08-27:
I know this issue is different from my problem, but it relates to parallel build.
Maybe instead of using `free` at line [88](https://github.com/m2-farzan/ros2-galactic-PKGBUILD/blob/extended/PKGBUILD#L88), better command will be: `grep MemTotal /proc/meminfo | grep -Po "MemTotal:\s+\K\d+"`.
Why? When your system has a different language system than the English `free` command returns translated value.

My output of `free` command:
``` sh
               razem       użyte       wolne    dzielone   buf/cache    dostępne
Pamięć:   32792452     3422640    22699636       56960     6670176    28847684
Wymiana:     2097148           0     2097148
```