m2-farzan/ros2-galactic-PKGBUILD #41: Using yay to install encountered an error

StudyingLover, 2023-08-11:
I‘m using `yay ros2-galactic` and meet a mistake 
```
Finished <<< pybind11_vendor [15.8s]
--- stderr: fastrtps                                                                 
In file included from /home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/src/cpp/dynamic-types/TypeNamesGenerator.cpp:1:
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:17:13: 错误：‘uint32_t’未声明
   17 |             uint32_t bound,
      |             ^~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:23:13: 错误：‘uint32_t’未声明
   23 |             uint32_t bound,
      |             ^~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:28:31: 错误：‘uint32_t’在此作用域中尚未声明
   28 |             const std::vector<uint32_t>& bound,
      |                               ^~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:8:1: 附注：‘uint32_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
    7 | #include <vector>
  +++ |+#include <cstdint>
    8 | 
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:28:39: 错误：模板第 1 个参数无效
   28 |             const std::vector<uint32_t>& bound,
      |                                       ^
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:28:39: 错误：模板第 2 个参数无效
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:33:31: 错误：‘uint32_t’在此作用域中尚未声明
   33 |             const std::vector<uint32_t>& bound,
      |                               ^~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:33:31: 附注：‘uint32_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:33:39: 错误：模板第 1 个参数无效
   33 |             const std::vector<uint32_t>& bound,
      |                                       ^
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:33:39: 错误：模板第 2 个参数无效
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:34:13: 错误：‘uint32_t’未声明
   34 |             uint32_t& ret_size,
      |             ^~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:40:13: 错误：‘uint32_t’未声明
   40 |             uint32_t bound,
      |             ^~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/src/cpp/dynamic-types/TypeNamesGenerator.cpp:11:13: 错误：no declaration matches ‘std::string eprosima::fastrtps::types::TypeNamesGenerator::get_string_type_name(uint32_t, bool, bool)’
   11 | std::string TypeNamesGenerator::get_string_type_name(
      |             ^~~~~~~~~~~~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:16:36: 附注：备选是： ‘static std::string eprosima::fastrtps::types::TypeNamesGenerator::get_string_type_name(int, bool, bool)’
   16 |     RTPS_DllAPI static std::string get_string_type_name(
      |                                    ^~~~~~~~~~~~~~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:13:7: 附注：‘class eprosima::fastrtps::types::TypeNamesGenerator’ defined here
   13 | class TypeNamesGenerator
      |       ^~~~~~~~~~~~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/src/cpp/dynamic-types/TypeNamesGenerator.cpp:26:13: 错误：no declaration matches ‘std::string eprosima::fastrtps::types::TypeNamesGenerator::get_sequence_type_name(const std::string&, uint32_t, bool)’
   26 | std::string TypeNamesGenerator::get_sequence_type_name(
      |             ^~~~~~~~~~~~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:21:36: 附注：备选是： ‘static std::string eprosima::fastrtps::types::TypeNamesGenerator::get_sequence_type_name(const std::string&, int, bool)’
   21 |     RTPS_DllAPI static std::string get_sequence_type_name(
      |                                    ^~~~~~~~~~~~~~~~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:13:7: 附注：‘class eprosima::fastrtps::types::TypeNamesGenerator’ defined here
   13 | class TypeNamesGenerator
      |       ^~~~~~~~~~~~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/src/cpp/dynamic-types/TypeNamesGenerator.cpp:42:13: 错误：no declaration matches ‘std::string eprosima::fastrtps::types::TypeNamesGenerator::get_array_type_name(const std::string&, const std::vector<unsigned int>&, bool)’
   42 | std::string TypeNamesGenerator::get_array_type_name(
      |             ^~~~~~~~~~~~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:31:36: 附注：备选为： ‘static std::string eprosima::fastrtps::types::TypeNamesGenerator::get_array_type_name(const std::string&, const int&, int&, bool)’
   31 |     RTPS_DllAPI static std::string get_array_type_name(
      |                                    ^~~~~~~~~~~~~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:26:36: 附注：         ‘static std::string eprosima::fastrtps::types::TypeNamesGenerator::get_array_type_name(const std::string&, const int&, bool)’
   26 |     RTPS_DllAPI static std::string get_array_type_name(
      |                                    ^~~~~~~~~~~~~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:13:7: 附注：‘class eprosima::fastrtps::types::TypeNamesGenerator’ defined here
   13 | class TypeNamesGenerator
      |       ^~~~~~~~~~~~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/src/cpp/dynamic-types/TypeNamesGenerator.cpp:51:13: 错误：no declaration matches ‘std::string eprosima::fastrtps::types::TypeNamesGenerator::get_array_type_name(const std::string&, const std::vector<unsigned int>&, uint32_t&, bool)’
   51 | std::string TypeNamesGenerator::get_array_type_name(
      |             ^~~~~~~~~~~~~~~~~~
/home/zjh/.cache/yay/ros2-galactic/src/ros2/src/eProsima/Fast-DDS/include/fastrtps/types/TypeNamesGenerator.h:31:36: 附注：备选为： ‘static std::string eprosima::fastrtps::types::TypeNamesGenerator::get_array_type_name(const std::string&, const int&, int&, bool)’
[... 28 lines cut ...]
      |                  ^~~~~~
make[1]: *** [CMakeFiles/Makefile2:923：src/cpp/CMakeFiles/fastrtps.dir/all] 错误 2
make: *** [Makefile:166：all] 错误 2
---
Failed   <<< fastrtps [22.4s, exited with code 2]
Aborted  <<< iceoryx_posh [31.2s]                   
Aborted  <<< yaml_cpp_vendor [37.4s]                
Aborted  <<< google_benchmark_vendor [43.5s]        
Aborted  <<< shared_queues_vendor [41.1s]                                                 
Aborted  <<< uncrustify_vendor [1min 27s]                                                     
Aborted  <<< rviz_ogre_vendor [6min 33s]                                            

Summary: 79 packages finished [6min 44s]
  1 package failed: fastrtps
  6 packages aborted: google_benchmark_vendor iceoryx_posh rviz_ogre_vendor shared_queues_vendor uncrustify_vendor yaml_cpp_vendor
  83 packages had stderr output: ament_clang_format ament_clang_tidy ament_cmake ament_cmake_auto ament_cmake_clang_format ament_cmake_clang_tidy ament_cmake_copyright ament_cmake_core ament_cmake_cppcheck ament_cmake_cpplint ament_cmake_export_definitions ament_cmake_export_dependencies ament_cmake_export_include_directories ament_cmake_export_interfaces ament_cmake_export_libraries ament_cmake_export_link_flags ament_cmake_export_targets ament_cmake_flake8 ament_cmake_gmock ament_cmake_gtest ament_cmake_include_directories ament_cmake_libraries ament_cmake_lint_cmake ament_cmake_mypy ament_cmake_nose ament_cmake_pclint ament_cmake_pep257 ament_cmake_pycodestyle ament_cmake_pyflakes ament_cmake_pytest ament_cmake_python ament_cmake_target_dependencies ament_cmake_test ament_cmake_version ament_cmake_xmllint a [... cut ...] z_ogre_vendor shared_queues_vendor sqlite3_vendor test_interface_files test_osrf_testing_tools_cpp tinyxml2_vendor tinyxml_vendor tracetools_read tracetools_trace uncrustify_vendor urdfdom_headers yaml_cpp_vendor zstd_vendor
  245 packages not processed
==> 错误： 在 build() 中发生一个错误。
    正在放弃...
 -> 生成时出错: ros2-galactic-exit status 4
 -> 无法安装以下软件包, 需要手动介入处理:
ros2-galactic - exit status 4
```

And I have installed fastrtps , but this problem is also existing .  How can I do to install ros2-galactic ?

awsms, 2023-09-27:
> This can be resolved by cherrypicking a patch for fast dds - as can be seen in #39:
> 
> ```
> ## eProsima Fast-DDS: fix missing cstdint include
> git -C $srcdir/ros2/src/eProsima/Fast-DDS cherry-pick -X theirs 02539b98a3a36e711da8f416372cbab2ecc178bb
> ```

why not creating a pull request to modify the PKGBUILD

m2-farzan, 2023-09-29:
@lprobsth If you know the fix, we'd appreciate it if you can make a PR. Please don't forget to update the pkgbuild source field & the .srcinfo.

awsms, 2023-09-29:
I've never been able to compile it, no idea where that tasks module should be coming from

```
==> Sources are ready.
==> Making package: ros2-galactic 2022.12.09-1 (Fri 29 Sep 2023 09:30:19 AM CEST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Removing existing $pkgdir/ directory...
==> Starting build()...
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> gtest_vendor
Starting >>> fastcdr
Starting >>> ament_cppcheck
Starting >>> iceoryx_utils
Starting >>> osrf_pycommon
Starting >>> osrf_testing_tools_cpp                                                                                                                                                        
Starting >>> google_benchmark_vendor
Starting >>> urdfdom_headers
Starting >>> ament_pycodestyle
Finished <<< gtest_vendor [1.40s]                                                                                                                                                                                                                             
Starting >>> gmock_vendor
Finished <<< google_benchmark_vendor [1.61s]                                                                                                                                                                                        
--- stderr: urdfdom_headers                                                                                                                                                                                                         
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

---
Finished <<< urdfdom_headers [1.74s]
--- stderr: ament_pycodestyle                                                                                                                                                                                                                                 
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/colcon_ros/task/ament_python/build.py", line 105, in build
    return await extension.build(additional_hooks=additional_hooks)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/colcon_core/task/python/build.py", line 81, in build
    available_commands = await self._get_available_commands(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/colcon_core/task/python/build.py", line 171, in _get_available_commands
    output = await check_output(
             ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/colcon_core/subprocess.py", line 128, in check_output
    assert not rc, f'Expected {args} to pass: {stderr_data}'
AssertionError: Expected ['/home/nop/.local/bin/python', 'setup.py', '--help-commands'] to pass: Traceback (most recent call last):
  File "/home/nop/.cache/yay/ros2-galactic/src/ros2/src/ament/ament_lint/ament_pycodestyle/setup.py", line 6, in <module>
    setup(
  File "/home/nop/.local/lib/python3.11/site-packages/setuptools/__init__.py", line 107, in setup
    return distutils.core.setup(**attrs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nop/.local/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 172, in setup
    ok = dist.parse_command_line()
         ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nop/.local/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 472, in parse_command_line
    if self.handle_display_options(option_order):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nop/.local/lib/python3.11/site-packages/setuptools/dist.py", line 1210, in handle_display_options
    return _Distribution.handle_display_options(self, option_order)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nop/.local/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 686, in handle_display_options
    self.print_commands()
[... 42 lines cut ...]
  File "/home/nop/.local/lib/python3.11/site-packages/setuptools/dist.py", line 1210, in handle_display_options
    return _Distribution.handle_display_options(self, option_order)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nop/.local/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 686, in handle_display_options
    self.print_commands()
  File "/home/nop/.local/lib/python3.11/site-packages/setuptools/dist.py", line 987, in print_commands
    cmdclass = ep.load()
               ^^^^^^^^^
  File "/usr/lib/python3.11/importlib/metadata/__init__.py", line 202, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
[... cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1140, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'tasks'

==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: ros2-galactic-exit status 4
 -> Failed to install the following packages. Manual intervention is required:
ros2-galactic - exit status 4
```

yameatmeyourdead, 2023-10-18:
Can you try install again with clean build, https://github.com/m2-farzan/ros2-galactic-PKGBUILD/pull/39 was just merged and yay works on my system.

yameatmeyourdead, 2023-10-19:
Can I have the output of `pacman -Q | grep python*` and `python3.11 -m pip freeze`?

awsms, 2023-10-19:
> Can I have the output of `pacman -Q | grep python*` and `python3.11 -m pip freeze`?

all python packages and dependencies are installed through pacman/yay
```
aiohttp==3.8.6
aiosignal==1.3.1
airdrop-ng==1.1
airgraph-ng==1.1
alabaster==0.7.13
annotated-types==0.5.0
ansible-compat==4.1.10
ansible-core==2.15.5
ansible-lint==6.21.1.dev0
anyio==3.6.2
anytree==2.10.0
apipkg==3.0.1
appdirs==1.4.4
apsw==3.43.1.0
arandr==0.1.11
argcomplete==3.0.8
args==0.1.0
asciidoc==10.2.0
asgiref==3.7.2
astroid==2.15.6
asttokens==2.4.0
async-generator==1.10
async-timeout==4.0.3
attrs==23.1.0
audiotools==3.1.1
autocommand==2.2.2
Automat==22.10.0
Babel==2.12.1
backcall==0.2.0
bcrypt==4.0.1
Beaker==1.12.1
beautifulsoup4==4.12.2
bencoder.pyx==3.0.1
binaryornot==0.4.4
black==23.7.0
blinker==1.6.2
boolean.py==4.0
bottle==0.12.25
bracex==2.4
Brotli==1.1.0
brotlicffi==1.0.9.2
bs4==0.0.1
btrfsutil==6.5.2
build==1.0.3
CacheControl==0.13.1
cachetools==5.3.1
cachy==0.3.0
catkin-pkg==1.0.0
cchardet==2.1.7
certifi==2023.7.22
cffi==1.16.0
chardet==5.2.0
charset-normalizer==3.3.0
cleo==2.0.1
cli-helpers==2.3.0
click==8.1.7
clikit==0.6.2
clint==0.5.1
clip==0.2.0
cloudscraper==1.2.71
colcon-argcomplete==0.3.3
colcon-bash==0.5.0
colcon-cd==0.1.1
colcon-cmake==0.2.28
colcon-common-extensions==0.3.0
colcon-core==0.15.0
[... 848 lines cut ...]
python-wcwidth 0.2.8-1
python-webcolors 1.13-2
python-webencodings 0.5.1-10
python-websocket-client 1.6.4-1
python-websockets 10.4-3
python-werkzeug 2.3.7-1
python-wheel 0.40.0-3
python-wrapt 1.14.1-2
python-wsproto 1.2.0-2
python-xapp 2.4.1-1
python-xlib 0.33-2
python-yaml 6.0.1-2
python-yarl 1.9.2-1
python-zeroconf 0.63.0-1
python-zipp 3.17.0-1
python-zope-event 5.0-1
python-zope-interface 6.0-3
python-zstandard 0.21.0-2
python2 2.7.18-8
python2-appdirs 1.4.4-6
python2-packaging 20.9-7
python2-pyparsing 2.4.7-6
python2-setuptools 2:44.1.1-2
python2-six 1.16.0-6
```